collin80/ESP32RET #3: Running on ESP32 Devboard?

jameswarner62, 2019-10-19:
Struggling to get this working on an ESP32 devkitC - has it been tested on this?

collin80, 2019-10-21:
I'm not sure but I believe it has. I use the actual EVTV boards but I compile as the ESP32 devkit so you'd think it would work. Perhaps recheck that you have everything connected to the proper pins. It's easy to make a mistake or wrong assumption there.

jameswarner62, 2019-10-21:
Thanks - will keep at it then

closed 2019-10-21 by jameswarner62

jameswarner62, 2019-10-21:
Found the issue for any other attempting this -  the USB>serial transceiver on the devboard does not support 1000000 baud - moving it back to 115200 in code and using savvycan via wifi works fine

EdKok, 2025-04-14:
Old, I know. But does anyone know where to change this in code?