collin80/ESP32RET #4: SavvyCan says connected but no CAN frames

fbettag, 2020-05-01:
So i have gotten me a SuperB for my Macchina M2. I flashed the Macchina M2 with the M2RET and using a USB Cable, everything works flawlessly.

I have modified the ESP32RET.ino so it runs as an AP with my configured name and psk. That works well, i can connect to the WiFi and SavvyCan also reports connected and shows me some kind of ping data on the console output.

However, i only get CAN frames when i am connected to the M2 via Cable, i am not getting anything when connected to WiFi on the SuperB.

Does the m2 need a different Firmware when i am using the SuperB? I am not quite sure what the problem here is.

Any help is appreciated.

collin80, 2020-05-03:
Unfortunately, there is a good chance that M2RET doesn't properly send CAN frames to the SuperB at this point. I had trouble with the SuperB I've got here and never got my hardware issues resolved so support is probably a bit lacking. I'll try to go back and correct the issues on my end then fix M2RET so that things are operational.

fbettag, 2020-05-03:
Thank you! I read an answer from the savvycan forums from you and tried to write my own esp-bridge which didn’t work out too well so far.

Thank you, can’t wait to get working with this :)

fbettag, 2020-05-03:
So from what i re-read just now, this is the wrong repo to open the issue ;) I will close this here then. Do you want me to reopen this issue on the m2ret repo?

closed 2020-05-03 by fbettag