collin80/ESP32RET #6: Wifi reliability

Hurde, 2021-01-11:
I've managed to get 3 devices working with esp32RET and 2 Due's using GVRET.
I am testing nearly all the devices on one car, started with the Dues.
The reliability of the Due's is really good I had no issues with them.

For the esp32 it is somewhat a different story.
I started a prototype board with the "ESP32 Devkit v1" which uses the ESP32-WROOM-32.
The distance from where I am standing to my car where the device is located is around 4,5 - 5 meters, which should be nothing for wifi.
It seems I was wrong, I had no signal at all from where I was standing.
So went into the car near the device and I managed to get a connection and started testing, which didn't fail and was quite reliable.
But the fact the wifi didn't reach that far was frustrating and seems to be a different issue for the board itself.

So the next board I tried was "TTGO LoRa32 v1 OLED" which was somewhere lying around at home.
This board had a nice wifi antenna on the bottom so I hoped it would perform better.
I managed to get the sketch working, it communicated on a test CAN networking using the two Due's and it worked.
But the antenna was near some pins from the board itself and I wasn't really satisfied after all before even testing in the car.

The last board I tried was "TTGO Tbeam v1" this board had the same antenna but freely located on the board and had better power supply options.
Then again I got the sketch working I only had to change the can pin to pin 4 & 0, pin 5 was used for a Lora module.
( This board has PSRAM as well but as said on the main page, I disabled it )
The wifi signal was significantly better than the other two boards.
This board went into the car and I tried it again from 4,5 - 5 meter.
Surprisingly the signal dropped enormously but I managed to set up a connection. 
Savvycan works but only for a minute on a loaded bus, then the bus starts to stutter by looking at the receiving frames and eventually results into "Connected to 0 busses" and the wifi dropped as well.

My actual question is: How is the wifi reliability for the Macchina A0 ? because I am trying to figure out what the problem could be.

collin80, 2021-01-12:
My experience is that the wifi performance seems quite good on the A0. I'm employed at EVTV where we also make a "Due" form factor ESP32 board and it has been designed to have good wifi performance too. Lots of people are using our ESP32 board for solar installations where the board communicates wirelessly to Raspberry Pi units used as displays. So, there do exist boards with decent wifi signal.

embenix, 2023-08-18:
> My experience is that the wifi performance seems quite good on the A0. I'm employed at EVTV where we also make a "Due" form factor ESP32 board and it has been designed to have good wifi performance [...]

Hi,
I am facing pretty much the same problem. I have tried your code with multiple ESP32 boards from different vendors mainly Adafruit Huzzah, Espressif Devkits and M5Stack Atom (works with WiFi only). All seems to connect and work for a few seconds, I see the frames coming for some time and then suddenly it stops. Console however, still shows data coming in. The only difference is the data size. While the Frames were successfully displayed on the software, the data length was fixed 120bytes, later this crosses 200 between 400 length.
It works every time when I reset the device but stops after a few seconds. Don't know where to look or what to fix??

Can you please help me in this regard? Thanks in Advance.

txfu, 2023-11-16:
> Hi, I am facing pretty much the same problem. I have tried your code with multiple ESP32 boards from different vendors mainly Adafruit Huzzah, Espressif Devkits and M5Stack Atom (works with WiFi [...]

Hi,
I am also experiencing very similar issues and symptoms to the above.
I am using a ESP32 board with CG340G as another board with CP2102 would not work correctly on the serial monitor @ 1000000Kpbs

ESP32RET successfully connects and receives frames as expected over WiFi. It will then work for an undetermined amount of time - sometimes a few seconds, sometimes a few minutes and disconnect. When it does fail, behaviour & console output is as follows:

- Working: Data length = 80
- Before Failure: Data length = 84, then 400, then 320
- Failure: "Write to Serial -> f1 f9"
- Failure: "Comms validation failed"

Bus impedance is 60ohms as suspected & wiring all checked. Confirmed no issues when using my USB CAN analyzer

Any help or suggestions would be much appreciated! Thank you