collin80/ESP32RET #10: EVTV ESP32 CANDue Board with Enclosure

hadlian, 2022-01-03:
I purchased this board in 2019 and tried to compile the ESP2RET program. In reading the readme file I changed the board as suggested. Below is the end of the compiling message.

Can someone assist on what I am missing. I have successfully loaded the companion CANTestESP32_both in the same IDE
Below are some of the details
Thanks

Arduino: 1.8.15 (Windows 10), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Using library WiFi at version 1.0 in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
Using library ESPmDNS at version 1.0 in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ESPmDNS 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ArduinoOTA 
Using library Update at version 1.0 in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Update 
Using library EEPROM at version 1.0.3 in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\EEPROM 
exit status 1
Error compiling for board ESP32 Dev Module.

collin80, 2022-01-03:
Hmmm, it's not really showing why there is an error. That's odd. Usually there's some good reason - missing library, wrong board selected, etc. But, I'm not seeing the actual cause here. It appears you have selected everything properly. It might be worth trying verbose compilation. Go to File->Preferences and then "Show verbose output during" check the compilation checkbox. This should create much more output. Perhaps it will help to shed light on what is happening.

hadlian, 2022-01-04:
Thanks for getting back to me so quickly.

bear with me I am not an expert but am starting to look at CAN data .
I have a winndows 7 machine that did compile  but wont connect.

It sees uno devices I must have done something wrong with the device driver. 
I will have to retrace steps.

Here is the Win10 messages.

You really help people like us out.
Do you have a patreon page?

Regards

> Hmmm, it's not really showing why there is an error. That's odd. Usually there's some good reason - missing library, wrong board selected, etc. But, I'm not seeing the actual cause here. It appears [...]

Arduino: 1.8.15 (Windows 10), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Harry\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Harry\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Harry\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6010 -ide-version=10815 -build-path C:\Users\Harry\AppData\Local\Temp\arduino_build_935701 -warnings=all -build-cache C:\Users\Harry\AppData\Local\Temp\arduino_cache_601209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Harry\AppData\Local\Temp\arduino_modified_sketch_535869\ESP32RET.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Harry\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Harry\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Harry\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=0403_6010 -ide-version=10815 -build-path C:\Users\Harry\AppData\Local\Temp\arduino_build_935701 -warnings=all -build-cache C:\Users\Harry\AppData\Local\Temp\arduino_cache_601209 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Harry\AppData\Local\Temp\arduino_modified_sketch_535869\ESP32RET.ino

Using board 'esp32' from platform in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Using core 'esp32' from platform in folder: C:\Users\Harry\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

cmd /c if exist "C:\\Users\\Harry\\AppData\\Local\\Temp\\arduino_modified_sketch_535869\\partitions.csv" copy /y "C:\\Users\\Harry\\AppData\\Local\\Temp\\arduino_modified_sketch_535869\\partitions.csv" "C:\\Users\\Harry\\AppData\\Local\\Temp\\arduino_build_935701\\partitions.csv"

cmd /c if not exist "C:\\Users\\Harry\\AppData\\Local\\Temp\\arduino_build_935701\\partitions.csv" copy "C:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\min_spiffs.csv" "C:\\Users\\Harry\\AppData\\Local\\Temp\\arduino_build_935701\\partitions.csv"

Detecting libraries used...

"C:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardwa [... cut ...] Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wifi_provisioning" "-IC:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\Harry\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-face" "-IC:\\Users\\Harry\\App

hadlian, 2022-01-14:
Hey Collin I got the ESP32 to compile and load on the EVTV CanDue device.

I open SavvyCan32 V199 and go to the connection page connect on com.

The device connects and there is activity on the boards leds but nothing comes to the application.

I have used another sniffer and the data is there.

Any suggestions?

Thank you