collin80/ESP32RET #11: missing the DLC byte for the ATH1 case.

kc427hk, 2022-04-30:
In file ELM327_Emulator.cpp, function "processCANReply()", when the H1 is used, ie, bHeader is true, it seems that DLC byte is not set into the output buffer.

collin80, 2022-05-02:
I believe the default behavior is still to have no DLC value unless you issue an ATD1 command. But, the firmware as it sits on github doesn't respond to ATD1 so it would never show the DLC. I've added support to my local copy. I should be uploading a new version in a couple of days.

kc427hk, 2022-05-03:
Just curious, is it ATD1 or ATH1?  I tried using Car Scanner, and and the firmware won't respond correctly because Car Scanner app issues ATH1.  However, it works find on Torque because it uses ATH0.  I have tired to added the DLC if ATH1 locally, and finally, Car Scanner app works fine.

Also, in your another repository file, https://github.com/collin80/GEVCU/blob/master/ELM327Processor.cpp, line 86, you also added DLC if ATH1 case, ie bHeader is true.