collin80/ESP32RET #13: Compile issue:

flyboy013, 2022-06-27:
I’m getting errors when attempting to compile ESP32RET.  Specifically, I’m getting:

```
wifi_manager.cpp:36:32: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
```

This error is in WiFiManger::setup(), which is where info is defined as type WiFiEventInfo_t.   Here is an excerpt of the code:

```
  WiFiEventId_t eventID = WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info) 
        {
           if (SysSettings.fancyLED)
           {
               leds[0] = CRGB::Red;
               FastLED.show();
           }
           Serial.print("WiFi lost connection. Reason: ");
           Serial.println(info.disconnected.reason);
           SysSettings.isWifiConnected = false;
```

ESP32RET includes WiFi.h which includes WiFiType.h which defines arduino_event_info_t as WiFiEventInfo_t.

WiFiEventInfo_t is defined in WiFiGeneric.h as show below.  There is no member disconnected.
```
typedef union {
        wifi_event_sta_scan_done_t wifi_scan_done;
        wifi_event_sta_authmode_change_t wifi_sta_authmode_change;
        wifi_event_sta_connected_t wifi_sta_connected;
        wifi_event_sta_disconnected_t wifi_sta_disconnected;
        wifi_event_sta_wps_er_pin_t wps_er_pin;
        wifi_event_sta_wps_fail_reason_t wps_fail_reason;
        wifi_event_ap_probe_req_rx_t wifi_ap_probereqrecved;
        wifi_event_ap_staconnected_t wifi_ap_staconnected;
        wifi_event_ap_stadisconnected_t wifi_ap_stadisconnected;
        wifi_event_ftm_report_t wifi_ftm_report;
        ip_event_ap_staipassigned_t wifi_ap_staipassigned;
        ip_event_got_ip_t got_ip;
        ip_event_got_ip6_t got_ip6;
        smartconfig_event_got_ssid_pswd_t sc_got_ssid_pswd;
        esp_eth_handle_t eth_connected;
        wifi_sta_config_t prov_cred_recv;
        wifi_prov_sta_fail_reason_t prov_fail_reason;
} arduino_event_info_t;
```

Fully compile output is:

```
Arduino: 1.8.16 (Mac OS X), Board: "Onehorse ESP32 Dev Module, 80MHz, 921600, None"

In file included from /Users/xxxx/Documents/Arduino/libraries/FastLED/src/FastLED.h:67,
                 from /Users/xxxx/Documents/Arduino/ESP32RET/ESP32RET.ino:32:
/Users/xxxx/Documents/Arduino/libraries/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/xxxx/Documents/Arduino/libraries/FastLED/src/FastLED.h:67,
                 from /Users/xxxx/Documents/Arduino/ESP32RET/wifi_manager.cpp:8:
/Users/xxxx/Documents/Arduino/libraries/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/xxxx/Documents/Arduino/ESP32RET/wifi_manager.cpp: In lambda function:
wifi_manager.cpp:36:32: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
            Serial.println(info.disconnected.reason);
                                ^~~~~~~~~~~~
                                eth_connected
wifi_manager.cpp:38:23: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
            if ( (info.disconnected.reason == 202) || (info.disconnected.reason == 3))
                       ^~~~~~~~~~~~
                       eth_connected
wifi_manager.cpp:38:60: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
            if ( (info.disconnected.reason == 202) || (info.disconnected.reason == 3))
                                                            ^~~~~~~~~~~~
                                                            eth_connected
/Users/xxxx/Documents/Arduino/ESP32RET/wifi_manager.cpp: In member function 'void WiFiManager::setup()':
wifi_manager.cpp:45:25: error: 'SYSTEM_EVENT_STA_DISCONNECTED' is not a member of 'arduino_event_id_t'
         }, WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: /Users/xxxx/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

What am I missing?  Has arduino_event_info_t changed such that member disconnected was removed?

Thx.

collin80, 2022-06-30:
Yes, recently they changed the names of some of the wifi status fields and I didn't update the code to match. So, if you (as I assume) are using the newest arduino-esp32 then compilation breaks like you saw. I just pushed a commit to fix this. Or, just comment out that whole block. It isn't strictly necessary to have that block but it does try to fix wifi connection problems.

flyboy013, 2022-06-30:
@collin80 Thank you.