collin80/ESP32RET #16: ESP32 dev board, connecting to savvyCAN but not giving frames

Lann4daze, 2022-12-04:


embenix, 2023-08-16:
I had the same problem for a long time and the only solution I could find was to use A0 system type.
`settings.systemType = 0;`
This fixed my problem and I started receive the messages on SavvyCAN software.
But now I have stuck to a different problem. SavvyCAN software stops updating the frames after some time. Time is not fixed and it can stop at any point. However, the console stays getting new messages over the serial port.

DNThomas, 2024-01-08:
Had the exact same issue. Just hacked up a bare ESP32 dev module with a transceiver IC and could connect with the same repeating "Got data from serial. Len = 4 f1 09 de ad" messages.

Looking at embenix's solution, all I could see is that currently, the system type 1 for the ESP doesn't explicitly set the CAN pins as the A0 version does.

I just added this line to the bottom of the ESP settings and it happily started loggin the bus:

        CAN0.setCANPins(GPIO_NUM_4, GPIO_NUM_5);

zygimantasmik, 2024-01-09:
Hello @DNThomas,
I have been facing probably the same issue as @Landennelson06 and described it here
https://github.com/collin80/ESP32RET/issues/26#issuecomment-1877309749

I am using systemType == 0 and the pins assignments are defined CAN0.setCANPins(GPIO_NUM_4, GPIO_NUM_5); but only thing I am getting is write to serial -> f1 9 Got data from serial. Len = 4 f1 09 de ad over and over again

I don't know if it is related issue but had problems when using ESP32 boards and CAN libraries that the CAN speed was half of set one. 

Board I am using is 
[image]

r00li, 2024-10-06:
Did anyone here manage to fix this issue on generic ESP32 boards?
I am using the systemType = 0, I removed the ELMxxxx emulation and changed the baud rate to 921600 (since the default doesn't work properly). I am trying both USB serial connection and wifi connection and I just can't get this to work. When using wifi I get "New client: 0 192.168.1.178" log in the serial console, and SavvyCAN says connected, but looking at the output I am getting the same errors above as others. Furthermore if I try to change the CAN bus speed from SavvyCAN the board crashes - setting it from the serial menu seems to work though.

sheedyaja, 2025-02-13:
any solution? Found some problem

MaydayDragal, 2025-02-15:
I have the same issue as well! Any progress on this topic?

VictorJJim, 2025-02-20:
Update, I've gotten my ESP32 30pin dev kit board to work. Here is what I did:
1. I set the systemtype=0
2. I connected to pins 16 and 17 (Labeled pins TX2 and RX2 on the board) NOT GPIO pins 4 and 5. I believe this was the true problem.

gitgrimbo, 2025-05-31:
Same issue with ESP32-WROOM-32 connected to a [MrDIY shield](https://store.mrdiy.ca/p/esp32-can-bus-shield/).

Seems to connect to savvyCAN ok but then loops with 

```
Write to serial -> f1 9 
Got data from serial. Len = 4
f1 09 de ad 
Write to serial -> f1 9 
Got data from serial. Len = 4
f1 09 de ad 
...
```

SYC-O, 2025-07-23:
I have the same issue with the ESP32 Wroom board with usb C and mrdiy.ca shield. Is there a fix for this?