collin80/ESP32RET #21: Compilation errors

Kabron287, 2023-02-25:
Arduino 1.8.13/VisualMicro 2021.917.5
Suddenly I began to get errors:

In file included from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can_builtin.h:35:0,
                 from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can.h:1,
                 from C:\Temp\arduino_build_544023\sketch\config.h:35,
                 from C:\Temp\arduino_build_544023\sketch\Logger.h:31,
                 from C:\Temp\arduino_build_544023\sketch\Logger.cpp:27:
D:\Arduino\Proj\libraries\can_common\src/can_common.h:93:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
D:\Arduino\Proj\libraries\can_common\src/can_common.h:120:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
In file included from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can_builtin.h:35:0,
                 from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can.h:1,
                 from C:\Temp\arduino_build_544023\sketch\config.h:35,
                 from C:\Temp\arduino_build_544023\sketch\lawicel.cpp:6:
D:\Arduino\Proj\libraries\can_common\src/can_common.h:93:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
D:\Arduino\Proj\libraries\can_common\src/can_common.h:120:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
In file included from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can_builtin.h:35:0,
                 from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can.h:1,
                 from C:\Temp\arduino_build_544023\sketch\config.h:35,
                 from C:\Temp\arduino_build_544023\sketch\commbuffer.h:3,
                 from C:\Temp\arduino_build_544023\sketch\commbuffer.cpp:1:
D:\Arduino\Proj\libraries\can_common\src/can_common.h:93:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
D:\Arduino\Proj\libraries\can_common\src/can_common.h:120:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
In file included from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can_builtin.h:35:0,
                 from D:\Arduino\Proj\libraries\esp32_can\src/esp32_can.h:1,
                 from C:\Temp\arduino_build_544023\sketch\config.h:35,
                 from C:\Temp\arduino_build_544023\sketch\gvret_comm.h:3,
                 from C:\Temp\arduino_build_544023\sketch\gvret_comm.cpp:5:
D:\Arduino\Proj\libraries\can_common\src/can_common.h:93:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
C:\Temp\arduino_build_544023\sketch\Logger.cpp: In static member function 'static void Logger::log(Logger::LogLevel, const char*, va_list)':
Logger.cpp:175:12: error: enumeration value 'Off' not handled in switch [-Werror=switch]
     switch (level) {
            ^
D:\Arduino\Proj\libraries\can_common\src/can_common.h:120:42: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
         const bool operator[]( int pos ) const
                                          ^
C:\Temp\arduino_build_544023\sketch\Logger.cpp: In static member function 'static void Logger::logMessage(const char*, va_list)':
Logger.cpp:264:85: error: conversion lacks type at end of format [-Werror=format=]
                 writeLen = sprintf((char*)&buffer[buffLen], "%l", va_arg(args, long));
                                                                                     ^
Logger.cpp:264:85: error: too many arguments for format [-Werror=format-extra-args]
Logger.cpp:264:85: error: conversion lacks type at end of format [-Werror=format=]
Logger.cpp:264:85: error: too many arguments for format [-Werror=format-extra-args]
C:\Temp\arduino_build_544023\sketch\lawicel.cpp: In member function 'void LAWICELHandler::handleLongCmd(char*)':
C:\Temp\arduino_build_544023\sketch\lawicel.cpp:89:29: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (outFrame.length < 0) outFrame.length = 0;
                             ^
C:\Temp\arduino_build_544023\sketch\lawicel.cpp:101:29: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (outFrame.length < 0) outFrame.length = 0;
                             ^
C:\Temp\arduino_build_544023\sketch\lawicel.cpp:79:10: warning: unused variable 'buff' [-Wunused-variable]
     char buff[80];
          ^
C:\Temp\arduino_build_544023\sketch\lawicel.cpp: In member function 'void LAWICELHandler::sendFrameToBuffer(CAN_FRAME&, int)':
C:\Temp\arduino_build_544023\sketch\lawicel.cpp:299:13: warning: unused variable 'writtenBytes' [-Wunused-variable]
     uint8_t writtenBytes;
[... 140 lines cut ...]
                 from D:\Arduino\Proj\esp32\CAN\ESP32RET1\ESP32RET\ESP32RET.ino:32:
D:\Arduino\Proj\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
C:\Temp\arduino_build_544023\sketch\wifi_manager.cpp: In lambda function:
wifi_manager.cpp:36:32: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
            Serial.println(info.wifi_sta_disconnected.reason);
                                ^
wifi_manager.cpp:38:23: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
            if ( (info.wifi_sta_disconnected.reason == 202) || (info.wifi_sta_disconnected.reason == 3)) 
                       ^
wifi_manager.cpp:38:69: error: 'union system_event_info_t' has no member named 'wifi_sta_disconnected'
            if ( (info.wifi_sta_disconnected.reason == 202) || (info.wifi_sta_disconnected.reason == 3)) 
                                                                     ^
C:\Temp\arduino_build_544023\sketch\wifi_manager.cpp: In member function 'void WiFiManager::setup()':
wifi_manager.cpp:45:12: error: 'ARDUINO_EVENT_WIFI_STA_DISCONNECTED' was not declared in this scope
         }, ARDUINO_EVENT_WIFI_STA_DISCONNECTED); 
            ^
C:\Temp\arduino_build_544023\sketch\wifi_manager.cpp:28:23: warning: unused variable 'eventID' [-Wunused-variable]
         WiFiEventId_t eventID = WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info) 
                       ^
D:\Arduino\Proj\esp32\CAN\ESP32RET1\ESP32RET\ESP32RET.ino: In function 'void loop()':
D:\Arduino\Proj\esp32\CAN\ESP32RET1\ESP32RET\ESP32RET.ino:321:10: warning: variable 'isConnected' set but not used [-Wunused-but-set-variable]
     bool isConnected = false;
          ^

collin80, 2023-02-26:
Well, there certainly are a number of errors in the output. 

error: 'ARDUINO_EVENT_WIFI_STA_DISCONNECTED' indicates that you're using the wrong version of arduino-esp32. You should be able to use the newest released version.

Then there are errors such as  "error: enumeration value 'Off' not handled in switch" which is annoying. The value Off is not handled precisely because the desired effect is for nothing to happen in that case. The code probably should have an empty case block just for completeness but triggering an error on that is not something that should happen in C++ (now, Rust on the other hand will always do so). This is an easy fix in code but should have been, at most, a warning.

The error in Logger about too many arguments is strange. That code has compiled for years. 

My guess is that VisualMicro is setting some extremely strict compile flags and the code is too sloppy to pass them. I always just used the Arduino IDE which also somewhat hid the avalanche of warnings. Well, I knew about some of them but none of them actually affect program performance and so fixing them has been a low priority.

Kabron287, 2023-03-02:
Seems **Options/Visual Micro/Compiler/Warnings false**
solved the problem

Kabron287, 2023-03-03:
Here we disscused the problem:
https://www.visualmicro.com/forums/YaBB.pl?num=1677756965/4#4

closed 2023-05-01 by collin80