collin80/ESP32RET #23: Diy hardware - pinout

Filo301, 2023-03-19:
Hi,

I want to do own, basic hardware with two can buses and Led.
I beliv3?e machina a0 uses gpio 4&5 for can bus, how to add eg. Mcp2515? Which pins i should use?

collin80, 2023-03-21:
ESP32 has only one canbus. Lots of ESP32 based devices have multiple buses but they do so by adding external CAN modules like the MCP2515. Which pins you use are up to you but you can look at the code to see which pins other people have used as that makes things easier. The A0 uses pins 4 and 5 for the built-in CAN on the ESP32. You probably want to stick to standard pins like that. For the MCP2515 you'll obviously be using SPI and then the big question is which pin to use for CS and the interrupt signal pin. The default is 5 for CS and 27 for the INT pin. You'll see an obvious problem. It's because I used pins 16 and 17 for the built-in CAN. So, keeping as close to what I did as possible, pins 16 and 17 are built-in can, 5 is CS and 27 is INT from the MCP2515 chip

Filo301, 2023-03-25:
Thank You! I didn't sure about used pins to connect external can controller. Eventually did something similar to macchina a0 because it's only need 3.3v (ws2812 works well with 3v3). Some day I'll also make module with two can buses but i don't wanna do it with old mcp2515 and due to ic shortage i can't buy mcp2517 (or original macchina a0)

closed 2023-05-01 by collin80