collin80/ESP32RET #24: is this still maintained?

Pacerino, 2023-04-27:
I'm wondering if this Repo is still maintained? There are a lot of open issues and this wont compile with a ESP32-S3.

collin80, 2023-05-01:
Yeah, I'm still here. Though, it hasn't been super actively maintained, that's true. The code is old enough that the S3 didn't exist back when it was written. And, yes, the S3 is not supported (yet). Really, the best approach is probably to rebase the esp32_can project around the now existing TWAI code in the ESP32 IDF. There too, such things did not exist when I first made the CAN library. Basically, the biggest issue here is that I wrote all this stuff many years ago when ESP32 development was still somewhat in its infancy and now time has marched on and this project (and esp32_can) have just not kept up. The goal was to support EVTV hardware with the project and that was a success. It all does work well on that hardware which uses the original ESP32. Now there will be an EVTV board with the S3 so that gives me the push I need to finally get back to this. Long story short, it's going to happen in the near future.

Pacerino, 2023-05-02:
Thank you very much for your answer! I am looking forward to the future!

closed 2023-05-02 by Pacerino