collin80/ESP32RET #26: Configuring hardware for GVRET node for SavvyCAN

poldim, 2023-09-10:
Hello,  I'm trying to use [LilyGO T-CAN485](https://github.com/Xinyuan-LilyGO/T-CAN485) to connect to a DC Battery CAN BUS.  I'm trying to figure out how to customize the ESP32RET.ino to work with this hardware.  They have an [example](https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/example/Arduino/CAN/config.h) showing the following pins:

```
PIN_5V_EN 16
CAN_TX_PIN 26
CAN_RX_PIN 27
CAN_SE_PIN 23
WS2812_PIN 4
```

So I've modified the ESP32RET.ino to force using selection `3` and added the following:
```
    if (settings.systemType == 3)
    {
        Logger::console("Running on LILLYGO CANBUS BOARD");
        canBuses[0] = &CAN0;
        canBuses[1] = &CAN1;
        // SysSettings.LED_CANTX = 255;
        SysSettings.LED_CANRX = 255;
        // SysSettings.LED_LOGGING = 255;
        // SysSettings.LED_CONNECTION_STATUS = 255;
        SysSettings.fancyLED = true;
        SysSettings.logToggle = false;
        SysSettings.txToggle = false;
        SysSettings.rxToggle = true;
        SysSettings.lawicelAutoPoll = false;
        SysSettings.lawicelMode = false;
        SysSettings.lawicellExtendedMode = false;
        SysSettings.lawicelTimestamping = false;
        SysSettings.numBuses = 2;
        SysSettings.isWifiActive = true;
        SysSettings.isWifiConnected = true;
        // strcpy(deviceName, EVTV_NAME);
        // strcpy(otaHost, "media3.evtv.me");
        // strcpy(otaFilename, "/esp32ret.bin");
        //reconfigure the two already defined CAN buses to use the actual pins for this board.
        CAN0.setCANPins(GPIO_NUM_27, GPIO_NUM_26); //rx, tx
        // CAN1.setINTPin(36);
        // CAN1.setCSPin(32);

        FastLED.addLeds<WS2812B, 4, RGB>(leds, 1).setCorrection( TypicalLEDStrip );
        FastLED.setBrightness(  20 );
        //With the board facing up and looking at the USB end the LEDs are 0 1 2 (USB) 3
        //can test LEDs here for debugging but normally leave first three off and set connection to RED.
        //leds[0] = CRGB::White;
        //leds[1] = CRGB::Blue;
        //leds[2] = CRGB::Green;
        //leds[0] = CRGB::Red;
        FastLED.show();
        pinMode(16, OUTPUT);  //CAN 5V EN
        pinMode(23, OUTPUT);  //CAN CHIP MODE SELECT (SN65HVD231 Rs pin): LOW = HIGH SPEED, HIGH = SLOPE CONTROL
        digitalWrite(16, HIGH);
        digitalWrite(23, LOW);
    }
```

I can successfully flash the LilyGo board and connect to the AP.  Here is the seriaI log:
```
Loading settings....
3

Running on LILLYGO CANBUS BOARD
Build number: 618
Enabled CAN0 with speed 500000
Enabled CAN1 with speed 500000
Free heap after setup: 98792
Done with init
```

However, after opening SavvyCAN and configuring GVRET I'm unable to get a connection established.

[image]

**Any advice on what I can check to get this successfully working?**

collin80, 2023-09-11:
Well, one thing you probably need to change is that you're allowing for two CAN buses but you only have one. Trying to access the second can bus is going to be problematic as it is looking for an MCP2517FD on SPI. It may be causing huge delays or crashes in trying to talk to a device that doesn't exist. Also, try receiving CAN right on the text interface first before trying with SavvyCAN. It will show frames on the text window if CAN traffic is coming in. Lastly, try "LOGLEVEL=0" to get debugging info and re-enable the multi-second start up delay so that you can get a good look at what it is saying as everything initializes.

zygimantasmik, 2024-01-04:
Hello @collin80,
SavvyCAN is a very nice project and I would like to test it but I am facing problems using ESP32. I have managed to flash ESP32 and connect to Savvy but CAN BUS is not working I am unable to send or receive CAN data. I have also tried ESP32 CAN examples no data is going out from ESP32. I have used logic analyzer and RX TX pins are not togling. Is there are chance that additional libraries used in ESP32RET project are not working properly on "Aliexpress" edition ESP32-WROOM board?

poldim, 2025-09-09:
Sorry, I couldn't with savvy can. I did get it working with esphome but have since removed it.