collin80/ESP32RET #32: ESP32RET giving odd issue

Alibro53, 2024-09-24:
Hi guys
I have been using Savvycan for a while now and find it absolutely brilliant and always reliable when using Arduino Due but I'm trying to set it up with an ESP32 and having issues.
I have several different ESP32's and went through all of them tonight installing the ESP32RET software using the GVRET firmware updater. The software installed no problem on all of them but only some would connect in SAVVYCAN. 
Of those that did connect they all had issues reading CAN from an old Landrover Ipack at 500k baud. They all showed the same odd behaviour in that they showed no CAN until I put my finger across pins 4 and 5. If I am touching the ESP32 I see CAN but as soon as I stop touching it the CAN stops.
I have checked the termination and the ipack is already terminated so I used a transceiver that has termination resulting in 60R across the CAN bus, I'm also using twisted wire and the same transceiver works perfectly with a Due so I don't think that is the issue.
This is the the same as the transceiver I'm using.
https://amzn.to/4gF68Ju

I also have a couple of these with the resistor cut out and found that it doesn't matter which one I use I get the same issue.
Is it the ESP32 that I'm using or an earthing fault or am I missing something here.

Thanks

collin80, 2024-09-26:
Well... I've never heard of needing to put your fingers onto the pins to get it working. Are you touching pins 4 and 5 of the ESP32 or the transceiver? By default, the ESP32 library uses pins 16 and 17 of the ESP32 for CAN. So, if you're using pins 4 and 5, maybe try different pins and see if things behave better.

Alibro53, 2024-09-26:
LOL, Thanks Collin, I'll try that tomorrow.