collin80/ESP32RET #42: Macchina A0 suddenly not working with ESP32RET

Ldalvik, 2025-03-24:
I have been using ESP32RET and SavvyCAN for the past few days with no issues. Then, I believe I updated my ESP32 board libraries. After testing multiple versions, (2.0.0, 2.0.17, 3.0.0, etc) I reduced it down to v3.0.0 getting me farthest. Other versions will throw weird errors like "spin lock assert" crashes.

With ESP32 board library 3.0.0, the serial output is 

"Loading settings...
Running on Macchina AO
TWAI driver installed
Creating queues
Starting can handler task"

And it stops there, LED staying red. I have been reverse engineering my CAN bus all week and the code still works to read RPM, current gear, etc. so it is not an issue with my CAN bus. I just cannot get ESP32RET to work anymore. Even after flashing the ESP32RET.bin, it says the board is an EVTV or something. Any help would be appreciated, as I feel like I've tried everything.

collin80, 2025-03-24:
On the savvycan website is a link to an already compiled ESP32RET firmware that should work:

https://www.savvycan.com/ESP32RET_Updater.zip

Otherwise, yes, there have been a few recent kinks that can kick up and make things work poorly. It should work to freshly download the source code and then use the ESP32-Arduino version v3 to compile.

Ldalvik, 2025-03-25:
Thank you. I tried using the updater, but it thinks my device is an EVTV.
This didn’t happen when using the source code originally. Do you know how
to fix this?

Ldalvik, 2025-03-27:
After testing some more, I'm able to send commands to serial and check the status. It says WiFi is enabled and everything, but no WiFi AP is being created

collin80, 2025-03-27:
Hmmm, and you're sure you set the wifi mode to create an AP? It defaults to connecting to an existing SSID. If so, I'll have to check tonight to see if I can figure out why it wouldn't be creating an AP if you asked it to. It certainly was working last I knew.

Ldalvik, 2025-03-28:
It says it's WIFIMODE=2 when I flash the binary, and when I uploaded the code previously with arduino, it defaulted to creating the WiFi AP as well. The second the code uploaded, I saw the CAN frames in the serial output and the SSID (A0RETSSID or whatever) would appear. I never had to adjust any default settings to get it working with SavvyCAN previously.

Ldalvik, 2025-04-01:
Have you had a chance to test the code on a Macchina A0? I'd like to make sure it is not a problem with my device.

collin80, 2025-06-06:
I wonder if it would help if you erased the whole board and started fresh. I just tested a board (not an A0 though) with the very latest esp32_can, can_common, arduino esp32 files V3.2 and I can get it to boot up and also create an AP for me to connect to. So, it seems like it still ought to work. But, maybe the wifi data has been corrupted or something on your board. You can erase the whole board and start over fresh by executing esptool from the command line (esptool erase_flash). Then try uploading esp32ret again. If that doesn't help then I'll have to see what else I can find.